tomeshnet/mesh-orange #2: No IR and DFS relating to Mesh Point operation

benhylau, 2017-02-23:
Probably should write up something about no IR, radar detection / DFS and setting the regulatory domain.

See:
http://superuser.com/questions/809282/wifi-5ghz-ap-mode-what-does-no-ir-means-and-can-i-bypass-it
http://pisarenko.net/blog/2015/05/05/updates-to-the-802-dot-11ac-guide/

hamishcoleman, 2017-02-24:
Can we really say much about this?  From a purely configuration standpoint, all we can really say is something like "be sure to select the correct country for your regulatory domain"

I guess there is still some value in briefly explaining things, but I wouldn’t want to get too deep into details

benhylau, 2017-02-25:
Once the image is available, maybe a few lines in the README to explain what is the "regulatory domain" and how it relates to IR and DFS bands. With respect to selection, I think these are the options?

- SSH in to manually select on first-run (I don't prefer since we can otherwise completely auto-configure the node by just powering it up)
- Pick up from a config file on first-run
- Auto-detect the selected mesh id on all the bands

hamishcoleman, 2017-02-25:
> References

Why not do all three?

(possibly as a future thing...)

Said config file could be where the current reg domain is persisted to, so
the second run and onwards can just load what it contains.  Which makes the
SSH case one of "ssh in, update the config, reload the config".

Leaving the final option of "auto" - taking liberties here with your
suggestion - the IEEE 802.11d standard allows access points to advertise
the reg domain that stations need to use.  So, there could be an auto
"look at the scan results, choose a best guess" first-run option