tomeshnet/mesh-orange #9: How to copy built image

benhylau, 2017-04-30:
From Linux:
```
$ cat build/sun8i-h2plus-orangepi-zero.raw > /dev/$SDCARD
```

From Windows (?):
https://riot.im/app/#/room/!yzLpKcnymfPsdJyjnB:tomesh.net/$1493519399489626cWGRX:matrix.org

Verify:
>one way to check - eject it and re-insert it, see if it has a filesystem on it with "/boot" now

*NOTE: We may be changing `.raw` to `.img`.*

hamishcoleman, 2017-04-30:
I think I was trying to be too smart with the linux command - you need root to do this, but the above command does not work with sudo.

Instead:

sudo dd if=build/sun8i-h2plus-orangepi-zero.raw of=/dev/$SDCARD

Also probably useful for documentation at this point is "how do I know what my $SDCARD is":

lsblk -d -o NAME,SIZE,LABEL

benhylau, 2017-05-01:
Are we switching to `.img`?

hamishcoleman, 2017-05-01:
sure, why not

darkdrgn2k, 2018-06-16:
I use Win32DiskImager for Windows works well

Instructions can be pulled from Raspberry pi site
https://www.raspberrypi.org/documentation/installation/installing-images/windows.md

Also Linux instructions
https://www.raspberrypi.org/documentation/installation/installing-images/linux.md

And MacOS for good measure
https://www.raspberrypi.org/documentation/installation/installing-images/mac.md