tomeshnet/mesh-orange #10: Unauthenticated packages

benhylau, 2017-05-16:
Error I got trying to `make` from Gallium OS Chromebook:

```
After this operation, 105 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  gcc-6-base libc6 libgcc1 mawk base-files sensible-utils debianutils libbz2-1.0 liblzma5 multiarch-support libpcre3 libselinux1 zlib1g libattr1
  libacl1 tar dpkg dash libtinfo5 bash libgpg-error0 libgcrypt20 liblz4-1 libsystemd0 bsdutils coreutils diffutils e2fslibs libaudit-common
  libcap-ng0 libaudit1 perl-base debconf libpam0g libsemanage-common libsepol1 libustr-1.0-1 libsemanage1 libdb5.3 libpam-modules-bin libpam-modules
  passwd libuuid1 libblkid1 libcomerr2 libss2 libfdisk1 libmount1 libncursesw5 libsmartcols1 libudev1 util-linux e2fsprogs findutils grep gzip
  hostname libpam-runtime login mount ncurses-bin sed libdebconfclient0 base-passwd init-system-helpers libc-bin ncurses-base sysvinit-utils
  libapparmor1 libcap2 dmsetup libdevmapper1.02.1 libcryptsetup4 libidn11 libip4tc0 libkmod2 libseccomp2 adduser systemd lsb-base libncurses5
  libprocps6 procps udev tzdata gpgv debian-archive-keyring kmod libssl1.0.2 libbsd0 libedit2 libkeyutils1 libkrb5support0 libk5crypto3 libkrb5-3
  libgssapi-krb5-2 openssh-client dhcpcd5
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
apt download failed. Exit value: 100
Makefile:58: recipe for target 'build/tags/multistrap-pre.armhf' failed
make: *** [build/tags/multistrap-pre.armhf] Error 100
```

I probably just need to do this in a Debian VM.

hamishcoleman, 2017-05-16:
Do you have the rest of the build log prior to this error?

It should be downloading a debian-archive-keyring deb file, which contains the authentication for the rest of the packages.

The multistrap tool used to build this relies on a full set of working dpkg utilities in the host environment - which is where you might find incompatibilities with Ubuntu (they dont always bother to pull in the new versions from Debian...)

benhylau, 2017-05-16:
Yes I figured it's something like that.

```
W: GPG error: http://cdn-fastly.deb.debian.org/debian stretch InRelease: The following signatures couldn't be verified because the public key is not a$
ailable: NO_PUBKEY 8B48AD6246925553  NO_PUBKEY 7638D0442B90D010
W: The repository 'http://httpredir.debian.org/debian stretch InRelease' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

hamishcoleman, 2017-05-16:
I guess it is reasonable for Ubuntu not to include the trusted keys for Debian ...  but that does make it more difficult to install things :-(

Theoretically it is possible to install that debian-archive-keyring deb manually in ubuntu and it will add the trust keys - so if you could try that, we could document that as a way to build on Ubuntu.

If that doesnt work, the next step is to get the .gpg file out of that debian keyring and use apt-key to add it to the local trust.  Fun times...

hamishcoleman, 2017-05-16:
Actually, come to think about it - this kind of keyring fiddling should be done for the armbian package downloads too - we are getting the uboot, kernel and modules from a repository with no local trust for its signing key.

benhylau, 2017-05-16:
Yea I can `make` after doing:
```
sudo apt-get install debian-archive-keyring
```

benhylau, 2017-05-16:
Actually it failed at the end:
```
lzma <build/sun8i-h2plus-orangepi-zero.cpio >build/sun8i-h2plus-orangepi-zero.lzma
cat build/debian.stretch.armhf.lzma build/sun8i-h2plus-orangepi-zero.lzma >build/sun8i-h2plus-orangepi-zero.initrd
mkimage -C lzma -A arm -T ramdisk -d build/sun8i-h2plus-orangepi-zero.initrd build/boot/uInitrd
Image Name:   
Created:      Tue May 16 00:38:23 2017
Image Type:   ARM Linux RAMDisk Image (lzma compressed)
Data Size:    35525438 Bytes = 34692.81 kB = 33.88 MB
Load Address: 00000000
Entry Point:  00000000
mkdir -p build/boot/dtb
Touching tag dtb
cp build/sun8i-h2plus-orangepi-zero/usr/lib/linux-image-dev-sun8i/sun8i-h2plus-orangepi-zero.dtb build/boot/dtb/sun8i-h2plus-orangepi-zero.dtb
truncate --size=$((0x200)) build/sun8i-h2plus-orangepi-zero.img  # skip past the MBR
date -u "+%FT%TZ " >>build/sun8i-h2plus-orangepi-zero.img         # add a build date
git describe --long --dirty >>build/sun8i-h2plus-orangepi-zero.img # and describe the repo
fatal: No names found, cannot describe anything.
Makefile:271: recipe for target 'build/sun8i-h2plus-orangepi-zero.img' failed
make: *** [build/sun8i-h2plus-orangepi-zero.img] Error 128
```

And the img file is tiny:
```
$ ls -lh build/
total 206M
drwxrwxr-x  3 benedict benedict 4.0K Mar 16 23:04 boot
drwxr-xr-x 19 root     root     4.0K Mar 16 23:03 debian
drwxr-xr-x 20 root     root     4.0K May 16 00:36 debian.stretch.armhf
-rw-rw-r--  1 benedict benedict  71M May 16 00:36 debian.stretch.armhf.cpio
-rw-rw-r--  1 benedict benedict  25M May 16 00:37 debian.stretch.armhf.lzma
-rw-rw-r--  1 benedict benedict  31M Mar 16 23:04 initrd.lzma
-rw-rw-r--  1 benedict benedict  113 May 16 00:34 mtoolsrc
drwxr-xr-x  9 root     root     4.0K Feb  1 21:04 sun8i-h2plus-orangepi-zero
-rw-rw-r--  1 benedict benedict  38M May 16 00:37 sun8i-h2plus-orangepi-zero.cpio
-rw-rw-r--  1 benedict benedict  534 May 16 00:38 sun8i-h2plus-orangepi-zero.img
-rw-rw-r--  1 benedict benedict  34M May 16 00:38 sun8i-h2plus-orangepi-zero.initrd
-rw-rw-r--  1 benedict benedict 9.7M May 16 00:38 sun8i-h2plus-orangepi-zero.lzma
drwxrwxr-x  2 benedict benedict 4.0K May 16 00:38 tags
```

hamishcoleman, 2017-05-16:
bah!
I forgot to push the tags a couple of weeks back.

do a pull and try again, please

benhylau, 2017-05-16:
Different error:

```
Image Name:   
Created:      Tue May 16 02:01:35 2017
Image Type:   ARM Linux RAMDisk Image (lzma compressed)
Data Size:    35539288 Bytes = 34706.34 kB = 33.89 MB
Load Address: 00000000
Entry Point:  00000000
mkdir -p build/boot/dtb
Touching tag dtb
cp build/sun8i-h2plus-orangepi-zero/usr/lib/linux-image-dev-sun8i/sun8i-h2plus-orangepi-zero.dtb build/boot/dtb/sun8i-h2plus-orangepi-zero.dtb
truncate --size=$((0x200)) build/sun8i-h2plus-orangepi-zero.img.tmp   # skip past the MBR
date -u "+%FT%TZ " >>build/sun8i-h2plus-orangepi-zero.img.tmp          # add a build date
git describe --long --dirty >>build/sun8i-h2plus-orangepi-zero.img.tmp # and describe the repo
truncate --size=$((0x2000)) build/sun8i-h2plus-orangepi-zero.img.tmp  # skip to correct offset for SPL
cat build/sun8i-h2plus-orangepi-zero/usr/lib/linux-u-boot-dev-orangepizero_5.25_armhf/u-boot-sunxi-with-spl.bin >>build/sun8i-h2plus-orangepi-zero.img.tmp              # add the SPL+uboot binary
MTOOLSRC=build/mtoolsrc mpartition -I z:
/bin/sh: 1: mpartition: not found
Makefile:271: recipe for target 'build/sun8i-h2plus-orangepi-zero.img' failed
make: *** [build/sun8i-h2plus-orangepi-zero.img] Error 127
```

hamishcoleman, 2017-05-16:
can you tell me what version of mtools is installed?

in my environment here I have mtools 4.0.17-1, which provides /usr/bin/mpartition - which is what is missing according to the above error

benhylau, 2017-05-16:
It was not installed. The img built after I installed mtools.

benhylau, 2017-05-19:
It is included in `build-depends` and I have no idea why it was absent on my system. Closing.

commit 2eb2df7e (2017-05-16): Document building on Ubuntu and clarify expected build environment (fixes #10)
closed 2017-05-19 by benhylau