tomeshnet/mesh-orange #12: cjdroute.conf file, CJDNS establishment, Noob question

Sultan91, 2017-06-08:
Hello, I've just built an image booted up my orangePi for the first time. Could you give some advice on how to connect to Hyperboria for instance, 'cause I do not see any cjdroute.conf configuration files or any others.

Thanks

benhylau, 2017-06-08:
Installation of cjdns on the image is still work in progress, but you can do it manually using this as an example? https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/scripts/install2#L80-L115

Basically:
1. Get git and nodejs (cjdns needs nodejs to build)
1. Clone cjdns
1. Build it
1. Copy over the systemd files
1. Reboot

As for connecting to Hyperboria, you'd need to add a peer. Public ones are here: https://peers.fc00.io

Sultan91, 2017-06-08:
Yes that's the thing. I did the ordinary cjdns installation from https://docs.meshwith.me/install/debian-jessie.html and other sites. Logs say that cjdns up and running (active state) however I cannot manage connection to any public peers or even the local ones in cjdroute.conf. (was inserting credentials in UDPinterface: connectTO and ETHinterface connectTO as well)

benhylau, 2017-06-09:
What does peerStats say? Does it at least see the peer entry?

Sultan91, 2017-06-09:
Yes i did reboot and restart it several times. "peerStats command not found" Could explain how to use that command correctly?)

Sultan91, 2017-06-09:
Maybe you have some working connections, so that i could check it?)

benhylau, 2017-06-09:
Hmm... check iptables to make sure the ports aren't blocked? I haven't tried cjdns with this image. Will check it out next week.

Sultan91, 2017-06-09:
No worries, i've found some working peers thanks

benhylau, 2017-06-09:
Oh you got it to work? It was an issue with the peers then?

Sultan91, 2017-06-09:
Yep! I was able to ping some hyperboria sites, However don't know how to configure browser (firefox) to be able to connect to the sites.

benhylau, 2017-06-09:
I assume that's Firefox on your laptop connected to the mesh orange AP? You should just type in the domain or IPv6 and it will work if the orange zero is configured properly.

Sultan91, 2017-06-09:
Ok I will try, thanks

Sultan91, 2017-06-13:
Could you explain the one thing I am struggling with:
I need to deploy local mesh network running cjdns on all pi zeros and one node having internet gateway. Apart from installed cjdns software what other configurations are required. (e.g. do we need actually establish wifi mesh and install cjdns on top of it?)

benhylau, 2017-06-13:
Your nodes will first need a way of communicating with each other regardless of internet gateway, via ad-hoc or 802.11s mesh point. Then cjdns will bind to those interfaces and route. So each Pi zero needs an adapter that supports one of those modes, such as the ones listed in the first step here: https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/README.md#set-up Many other devices support these modes and you can find them with `iw list`.

Next you need to set up one internet gateway node (assigns IP and pins clients' cjdns public keys to allow exiting through) and configure each client to exit through the gateway (pin gateway node's cjdns public key in the IP tunnel section). There is some information and links here: https://github.com/tomeshnet/mesh-isp

darkdrgn2k, 2018-01-24:
Is this still an open issue?

benhylau, 2018-01-24:
Yes because cjdns is not bundled in, until we sort out the .deb install. Working on it at the moment.