tomeshnet/mesh-orange #16: hostapd and dns

darkdrgn2k, 2018-01-09:
Connection to HostAP establishes fine, but no DNS servers are offered over dhcp.

darkdrgn2k, 2018-01-09:
[image]

benhylau, 2018-01-09:
Confirmed this issue on MacOS

hamishcoleman, 2018-01-12:
systemd-networkd has a race condition - the dhcp service is configured before the uplink port is fully setup.  Thus the dhcp service does not see any DNS (or NTP) settings when it is first run.

Additionally, the version of systemd-networkd in stretch looks like it will only reconfigure the dhcp server if you restart the whole daemon - which just causes the same race.

Finally, a fix included in systemd v234 could help (see https://github.com/systemd/systemd/pull/6116) by allowing a down/up on the downstream interface to trigger a reconfigure of the dhcp service

benhylau, 2018-01-12:
Let's grab systemd v236 from `stretch-backports`?