tomeshnet/mesh-orange #19: Build is broke

benhylau, 2018-01-16:
Current `master` is broken. Building for both Raspberry Pi 2 and Orange Pi Zero will hang on this line:

```
...
mkdir -p build/debian.stretch.armhf/dev
sudo mknod build/debian.stretch.armhf/dev/urandom c 1 9
sudo mkdir -p build/debian.stretch.armhf/usr/sbin/
sudo cp policy-rc.d build/debian.stretch.armhf/usr/sbin/policy-rc.d
```

Head of `confd` branch works properly.

hamishcoleman, 2018-01-16:
fixed in my branch

benhylau, 2018-01-16:
After merging your branch, I have to first run `git submodule update --init` before calling `make build-depends`. Shall we add this to the root Makefile?

darkdrgn2k, 2018-01-17:
Issue was introduced in commit #[87a100a](https://github.com/tomeshnet/mesh-orange/commit/509b9e525f7a48e0ab85e8df7a20922b387a100a)

```
509b9e525f7a48e0ab85e8df7a20922b387a100a is the first bad commit
commit 509b9e525f7a48e0ab85e8df7a20922b387a100a
Author: Hamish Coleman <hamish@zot.org>
Date:   Fri Jan 12 16:54:47 2018 +0800

    Add remaining parts needed to implement debian build config overlays
```

Seems when  param --preserve-env was added breaks things in debain/Makefile

That directory now resides in another repo

[PR Created](https://github.com/hamishcoleman/debian-minimal-builder/pull/1)

hamishcoleman, 2018-01-17:
The error messages you dont recall seeing before have always been there - I just removed a lot of the extra messages hiding them.  They are benign.

The sudo issue is more concerning.  Please report what sudo version you are seeing this with - I am using sudo 1.8.21p2-2 and have no issues with these commands

hamishcoleman, 2018-01-17:
The "--preserve-env=$LIST" feature was added in sudo 1.8.21 (which was added to Debian in 2017-08-28)

I have replaced it with a less bleeding-edge construct in pull request #20

darkdrgn2k, 2018-01-17:
:+1:  Knew you would have a better solution

Clean install of Ubuntu 16.04.3 LTS just last week
```
root@userPC:~/mesh/mesh-orange# uname -a
Linux userPC 4.10.0-28-generic #32~16.04.2-Ubuntu SMP Thu Jul 20 10:19:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
root@userPC:~/mesh/mesh-orange# sudo --version
Sudo version 1.8.16
```

darkdrgn2k, 2018-01-17:
> I have replaced it with a less bleeding-edge construct in pull request #20

> I think the actual fix you are looking for is in the mesh-orange repo already as commit 9f3f0e7be96b13e250b9349b2e1287f545bdc7e7

Am I missing something. 
Checked out hamish20180117 to test same errors

```
Touching tag multistrap-post.armhf
Touching tag multistrap.armhf
sudo --preserve-env=CONFIGDIRS ./packages.addextra build/debian.stretch.armhf armhf minimise
sudo: option '--preserve-env' doesn't allow an argument
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p prompt] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p prompt] [-u user] file ...
```

darkdrgn2k, 2018-01-17:
do you need to specify only a single env variable? Is there harm in preserving them all ( --preserve-env=CONFIGDIRS  vs --preserve-env )

benhylau, 2018-01-17:
#20 fixes the build for me

@darkdrgn2k I changed the Vagrant's mesh-orange-images Makefile to clone from `hamishcoleman/mesh-orange` and `TAG = hamish20180117` and it worked. 
Perhaps your submodule is out of sync on the old hash?

darkdrgn2k, 2018-01-17:
Strange, I can now see the changes to those lines when I Browse the PR. Not sure why i didn't see them earlier.

Thanks

benhylau, 2018-01-17:
Perhaps you looked at it before @hamishcoleman pushed the commit? Closing this, thanks for testing :)

closed 2018-01-17 by benhylau