tomeshnet/mesh-orange #21: Raspberry Pi 3 support

benhylau, 2018-01-18:
Would like to use mesh-orange for a [workshop](https://github.com/tomeshnet/tpl-workshop/pull/3/files#diff-3148c32f6dbcfc89a7cb2667465362ecR32). What would it take to support the board?

hamishcoleman, 2018-01-19:
Have you tried it?

The notes in the board makefile claim that it works with rpi3

benhylau, 2018-01-19:
Yea there is no host ap and I cannot ssh to the Pi 3.

hamishcoleman, 2018-01-19:
You will need a serial console to get the debugging information.

You could also try checking out commit 9723ecaaad8fe0e8f08a82d66343275a1cd77d0f as I say there that I tested it and confirm it was working

benhylau, 2018-01-22:
I cannot build that branch at all. @darkdrgn2k think you may be able to check the output with serial console from the current master-built mesh-orange for Raspberry Pi 2 running on Raspberry Pi 3?

darkdrgn2k, 2018-01-23:
I was able to build master rpi2 and get it booted no problem.

Seems there are firmware files missing for wifi
[   51.185680] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2

[   49.691315] usb 1-1.4: Direct firmware load for ath9k_htc/htc_9271-1.4.0.fw failed with error -2

hamishcoleman, 2018-01-23:
I dont know if I ever tested the buitin wifi on the raspberry pi 3.  Sometimes the firmware messages you see are false leads - not all hardware needs all the firmware that it creates logs for.

So, you booted it up on a rpi3?  and got console log messages?  Did you get a working ethernet port?  When you logged in, what interfaces were discovered?  What state were they in?

I don't actually have any raspberry pi hardware, so I will not be able to replicate anything myself.

darkdrgn2k, 2018-01-23:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.35-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1014 SMP Fri Jun 30 14:47:43 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3d800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 14 pages/cpu @bcef8000 s25600 r8192 d23552 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 251720
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 elevator=deadline
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 933508K/1015808K available (7168K kernel code, 484K rwdata, 2012K rodata, 1024K init, 778K bss, 74108K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbe000000   ( 992 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c79014   ( 485 kB)
[    0.000000]        .bss : 0x80c7b000 - 0x80d3da64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000338] Console: colour dummy device 80x30
[    0.001228] console [tty1] enabled
[    0.001286] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001356] pid_max: default: 32768 minimum: 301
[    0.001717] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001760] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002857] Disabling cpuset control group subsystem
[    0.003046] CPU: Testing write buffer coherency: ok
[    0.003115] ftrace: allocating 22390 entries in 66 pages
[    0.055600] CPU0: update cpu_capacity 1024
[    0.055668] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.055765] Setting up static identity map for 0x100000 - 0x100034
[    0.058025] CPU1: update cpu_capacity 1024
[    0.058032] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.058752] CPU2: update cpu_capacity 1024
[    0.058758] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.059425] CPU3: update cpu_capacity 1024
[    0.059432] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.059515] Brought up 4 CPUs
[    0.059687] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.059716] CPU: All CPU(s) started in HYP mode.
[    0.059742] CPU: Virtualization extensions available.
[    0.060503] devtmpfs: initialized
[    0.071349] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.071662] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.071724] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.072287] pinctrl core: initialized pinctrl subsystem
[    0.073211] NET: Registered protocol family 16
[    0.075372] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.084117] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.084165] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.084338] Serial: AMBA PL011 UART driver
[    0.086240] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.086785] uart-pl011 3f201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.087148] irq: no irq domain found for /soc/aux@0x7e215000 !
[    0.156202] bcm2835-dma 3f007000.dma: DMA legacy API manager at be80f000, dmachans=0x1
[... 150 lines cut ...]
[   48.880168] systemd[1]: Reached target Remote File Systems.
[   48.894414] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[   48.909094] systemd[1]: Listening on udev Kernel Socket.
[   48.918260] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[   49.177533] systemd-journald[128]: Received request to flush runtime journal from PID 1
[... cut ...]
[   51.217626] usb 1-1.4: Direct firmware load for htc_9271.fw failed with error -2
[   51.217639] usb 1-1.4: no suitable firmware found!
[   51.217645] usb 1-1.4: ath9k_htc: Failed to get firmware htc_9271.fw
[   51.217689] usbcore: registered new interface driver ath9k_htc
[   51.218184] usb 1-1.4: ath9k_htc: USB layer deinitialized
[   52.207253] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   53.227348] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

Debian GNU/Linux 9 ramdisk-ddcad011 ttyS0

ramdisk-ddcad011 login:
```

darkdrgn2k, 2018-01-23:
Only the Raspberry pi 3 has a onboard wireless
eth0 interface up no problem

hamishcoleman, 2018-01-23:
What is the output of:

'''
ip link show
networkctl
networkctl show wlan0
ps afx
lsmod
'''

thanks

darkdrgn2k, 2018-01-23:
FYI we dealt with the ath9k issue back on [sept 15](https://riot.im/app/#/room/!FsFLbKGMcUXEMBxZdu:tomesh.net/$15054474698559vYXMD:tomesh.net) for the opi, im wondering if you just need to add the blob for that one.

```
login as: root
root@192.168.40.213's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
root@ramdisk-0f991711:~# ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:5f:ff:3c brd ff:ff:ff:ff:ff:ff
root@ramdisk-0f991711:~# networkctl
IDX LINK             TYPE               OPERATIONAL SETUP
  1 lo               loopback           carrier     unmanaged
  2 eth0             ether              routable    configured

2 links listed.
root@ramdisk-0f991711:~# networkctl show wlan0
Unknown operation show.
root@ramdisk-0f991711:~# ps afx
  PID TTY      STAT   TIME COMMAND
    2 ?        S      0:00 [kthreadd]
    3 ?        S      0:00  \_ [ksoftirqd/0]
    4 ?        S      0:00  \_ [kworker/0:0]
    5 ?        S<     0:00  \_ [kworker/0:0H]
    6 ?        S      0:00  \_ [kworker/u8:0]
    7 ?        S      0:00  \_ [rcu_sched]
    8 ?        S      0:00  \_ [rcu_bh]
    9 ?        S      0:00  \_ [migration/0]
   10 ?        S<     0:00  \_ [lru-add-drain]
   11 ?        S      0:00  \_ [cpuhp/0]
   12 ?        S      0:00  \_ [cpuhp/1]
   13 ?        S      0:00  \_ [migration/1]
   14 ?        S      0:00  \_ [ksoftirqd/1]
   15 ?        S      0:00  \_ [kworker/1:0]
   16 ?        S<     0:00  \_ [kworker/1:0H]
   17 ?        S      0:00  \_ [cpuhp/2]
   18 ?        S      0:00  \_ [migration/2]
   19 ?        S      0:00  \_ [ksoftirqd/2]
   20 ?        S      0:00  \_ [kworker/2:0]
   21 ?        S<     0:00  \_ [kworker/2:0H]
   22 ?        S      0:00  \_ [cpuhp/3]
   23 ?        S      0:00  \_ [migration/3]
   24 ?        S      0:00  \_ [ksoftirqd/3]
   25 ?        S      0:00  \_ [kworker/3:0]
   26 ?        S<     0:00  \_ [kworker/3:0H]
   27 ?        S      0:00  \_ [kdevtmpfs]
   28 ?        S<     0:00  \_ [netns]
   29 ?        S      0:00  \_ [khungtaskd]
   30 ?        S      0:00  \_ [oom_reaper]
   31 ?        S<     0:00  \_ [writeback]
   32 ?        S      0:00  \_ [kcompactd0]
   33 ?        S<     0:00  \_ [crypto]
   34 ?        S<     0:00  \_ [bioset]
   35 ?        S<     0:00  \_ [kblockd]
   36 ?        S<     0:00  \_ [watchdogd]
   37 ?        S      0:00  \_ [kworker/0:1]
   38 ?        S<     0:00  \_ [rpciod]
   39 ?        S<     0:00  \_ [xprtiod]
   40 ?        S      0:00  \_ [kworker/2:1]
   41 ?        S      0:01  \_ [kworker/1:1]
   42 ?        S      0:00  \_ [kworker/3:1]
   43 ?        S      0:00  \_ [kworker/u8:1]
   45 ?        S      0:00  \_ [kswapd0]
[... 54 lines cut ...]
  217 ?        Ss     0:00 /lib/systemd/systemd-networkd
  222 ?        Ss     0:00 /lib/systemd/systemd-resolved
  226 ttyS0    Ss+    0:00 /sbin/agetty --keep-baud 115200,38400,9600 ttyS0 vt220
  227 tty1     Ss+    0:00 /sbin/agetty --noclear tty1 linux
  241 ?        Ss     0:00 /usr/sbin/mactelnetd -n
root@ramdisk-0f991711:~# lsmod
Module                  Size  Used by
brcmfmac              222874  0
brcmutil                9092  1 brcmfmac
bcm2835_gpiomem         3940  0
uio_pdrv_genirq         3923  0
uio                    10204  1 uio_pdrv_genirq
fixed                   3285  0
ath9k_htc              61333  0
ath9k_common           29430  1 ath9k_htc
ath9k_hw              451961  2 ath9k_htc,ath9k_common
ath                    23073  3 ath9k_htc,ath9k_hw,ath9k_common
mac80211              655480  1 ath9k_htc
cfg80211              543027  5 ath9k_htc,mac80211,ath,brcmfmac,ath9k_common
rfkill                 20851  2 cfg80211
ip_tables              13161  0
x_tables               20578  1 ip_tables
ipv6                  408971  26
root@ramdisk-0f991711:~#
```

darkdrgn2k, 2018-01-23:
scp a copy of /lib/firmware/htc_9271.fw from an armbian box to /lib/firmware and tplink goes up

```
[ 1004.919378] usb 1-1.4: ath9k_htc: USB layer deinitialized
[ 1018.270929] usb 1-1.4: USB disconnect, device number 6
[ 1020.618673] usb 1-1.4: new high-speed USB device number 7 using dwc_otg
[ 1020.769601] usb 1-1.4: New USB device found, idVendor=0cf3, idProduct=9271
[ 1020.779111] usb 1-1.4: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[ 1020.791619] usb 1-1.4: Product: USB2.0 WLAN
[ 1020.798294] usb 1-1.4: Manufacturer: ATHEROS
[ 1020.805092] usb 1-1.4: SerialNumber: 12345
[ 1020.812684] usb 1-1.4: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[ 1020.826408] usb 1-1.4: Direct firmware load for ath9k_htc/htc_9271-1.4.0.fw failed with error -2
[ 1020.840558] usb 1-1.4: ath9k_htc: Firmware htc_9271.fw requested
[ 1021.143440] usb 1-1.4: ath9k_htc: Transferred FW: htc_9271.fw, size: 50980
[ 1021.391683] ath9k_htc 1-1.4:1.0: ath9k_htc: HTC initialized with 33 credits
[ 1024.893818] ath9k_htc 1-1.4:1.0: ath9k_htc: FW Version: 1.3
[ 1024.902154] ath9k_htc 1-1.4:1.0: FW RMW support: Off
[ 1024.909737] ath: EEPROM regdomain: 0x8348
[ 1024.909742] ath: EEPROM indicates we should expect a country code
[ 1024.909749] ath: doing EEPROM country->regdmn map search
[ 1024.909776] ath: country maps to regdmn code: 0x3a
[ 1024.909782] ath: Country alpha2 being used: US
[ 1024.909786] ath: Regpair used: 0x3a
[ 1024.926593] ieee80211 phy2: Atheros AR9271 Rev:1
[ 1025.690953] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1026.652035] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1032.541870] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

copied the /lib/firmware/brcm folder from raspbian and brcmfmac came up
```
[  946.215092] usbcore: deregistering interface driver brcmfmac
[  948.257256] usbcore: registered new interface driver brcmfmac
[  948.355697] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[  948.504979] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  948.514238] brcmfmac: power management disabled
[  948.769379] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
```

darkdrgn2k, 2018-01-23:
i may only need only two files

brcmfmac43430-sdio.bin  and brcmfmac43430-sdio.txt

hamishcoleman, 2018-01-23:
There is some boilerplate for adding firmware, so perhaps that just needs to be improved - I had thought that the ath9k had been added globally with that boilerplate.

The same logic goes for the raspberry pi stuff - it /should/ be simple to just tell the builder to add those files (in reality, it might be broken or need improvement)

I probably wont even be able to do a trial build until next week

darkdrgn2k, 2018-01-23:
Can you point us to the boilerplate please. What folder?

Raspberry-firmware\armbian-firmware

hamishcoleman, 2018-01-23:
the boilerplate is in the firmware directory.

The onboard wifi is probably as simple as determining which raspbian package contains those firmware files and then adding that name to the RASPBIAN_FIRMWARE_PKG line in the firmware/Makefile

The ath9k firmware is probably included in the armbian firmware package, so that might be a little more difficult - if it is simple, then it could be that just adding "include $(TOP_DIR)/boards/common-firmware-armbian.mk" to the boards/raspberrypi2/Makefile fie will bring that in.

(The firmware stuff is in need of refactoring - it currently includes far more than it should and has conflict issues)

darkdrgn2k, 2018-01-23:
Ah I see thanks. Ok I seeked out and added the two packages that are needed to rasbian install just to get it working ( firmware-brcm80211 and firmware-atheros )

Committed them to firmware branch if you want to test them out
@benhylau  if you want to try it build if checkout firmware, the tp link and the onboard wifi should work (the chineese board firmware was already there)

Lots of unneeded files
@hamishcoleman  Any suggestions how to delete the files you don't need

Note:  with onboard config they are both starting up as access points 
TPLink is not defined as meshable in  85-wlan-mesh.??? is is only "Driver=rt2800usb"

```
phy#0
        Interface wlan1
                ifindex 4
                wdev 0x1
                addr 18:d6:c7:0d:8a:86
                ssid test2
                type AP
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 30.00 dBm
phy#1
        Interface wlan0
                ifindex 3
                wdev 0x100000001
                addr b8:27:eb:0a:5a:69
                ssid test2
                type AP
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 31.00 dBm
```

darkdrgn2k, 2018-01-23:
Ref #22

benhylau, 2018-01-24:
Thanks @darkdrgn2k :) Since the patch is working, I am closing this issue so we continue this discussion on the PR.

closed 2018-01-24 by benhylau