tomeshnet/mesh-orange #27: Missing ath9k_htc on sun8i-h2-plus-orangepi-zero

benhylau, 2018-02-06:
Works on raspberrypi3 but not on sun8i-h2-plus-orangepi-zero:

>[  158.290385] usb 3-1: new high-speed USB device number 2 using ehci-platform
[  158.501113] usb 3-1: New USB device found, idVendor=0cf3, idProduct=9271
[  158.507829] usb 3-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[  158.515251] usb 3-1: Product: USB2.0 WLAN
[  158.519264] usb 3-1: Manufacturer: ATHEROS
[  158.523381] usb 3-1: SerialNumber: 12345
[  158.564330] usb 3-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[  158.571740] usbcore: registered new interface driver ath9k_htc
[  158.577691] usb 3-1: Direct firmware load for ath9k_htc/htc_9271-1.4.0.fw failed with error -2
[  158.586336] usb 3-1: Falling back to user helper
[  158.602860] usb 3-1: ath9k_htc: Firmware htc_9271.fw requested
[  158.608778] usb 3-1: Direct firmware load for htc_9271.fw failed with error -2
[  158.616118] usb 3-1: Falling back to user helper
[  158.632714] usb 3-1: no suitable firmware found!
[  158.637367] usb 3-1: ath9k_htc: Failed to get firmware htc_9271.fw
[  158.644108] usb 3-1: ath9k_htc: USB layer deinitialized

hamishcoleman, 2018-02-16:
have a look at the firmware/Makefile
There should not be two different lists of packages, but there is currently, so you could add the missing package name to the other list

benhylau, 2018-03-30:
Tried adding `firmware-atheros` but the package is not found during build:
https://github.com/benhylau/mesh-orange/blob/opi-atheros/firmware/Makefile#L10

Built with `armbian-firmware-full` but the package is 60+ MB, resulting in an image that does not boot on the Orange Pi Zero 512 MB.