tomeshnet/mesh-orange #32: Raspberry Pi 2

darkdrgn2k, 2018-02-28:
It seems the Raspberry Pi 2 is not executing `/init`

i tried booting it with `init=/init`  as a command line argument for the kernel and that didn't seem to work either.

hamishcoleman, 2018-03-01:
What does "git describe" tell you your repository version is?
Do you have a serial console log of a failed boot?
Does booting it with "init=/bin/bash" work?

darkdrgn2k, 2018-03-01:
I'm using @benhylau  build for the workshop (that works fine on a pi 3)
I also pulled a new mesh-orange and built it from scratch with same results

Does not seem anything is failing at boot

Regular boot
```[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.9.35-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1014 SMP Fri Jun 30 14:47:43 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3d800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 14 pages/cpu @bcef9000 s25600 r8192 d23552 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 251720
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 elevator=deadline
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 932468K/1015808K available (7168K kernel code, 484K rwdata, 2012K rodata, 1024K init, 778K bss, 75148K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbe000000   ( 992 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c79014   ( 485 kB)
[    0.000000]        .bss : 0x80c7b000 - 0x80d3da64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000011] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000034] Switching to timer-based delay loop, resolution 52ns
[    0.000389] Console: colour dummy device 80x30
[    0.001352] console [tty1] enabled
[    0.001411] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001488] pid_max: default: 32768 minimum: 301
[    0.001933] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001985] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003419] Disabling cpuset control group subsystem
[    0.003691] CPU: Testing write buffer coherency: ok
[    0.003787] ftrace: allocating 22390 entries in 66 pages
[    0.059174] CPU0: update cpu_capacity 1024
[    0.059252] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.059352] Setting up static identity map for 0x100000 - 0x100034
[    0.062558] CPU1: update cpu_capacity 1024
[    0.062570] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.063641] CPU2: update cpu_capacity 1024
[    0.063651] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.064625] CPU3: update cpu_capacity 1024
[    0.064634] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.064749] Brought up 4 CPUs
[    0.064950] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.064983] CPU: All CPU(s) started in HYP mode.
[    0.065013] CPU: Virtualization extensions available.
[    0.066090] devtmpfs: initialized
[    0.079226] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.079671] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.079752] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.080566] pinctrl core: initialized pinctrl subsystem
[    0.081896] NET: Registered protocol family 16
[    0.084370] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.094760] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.094821] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 336 lines cut ...]
[   57.030573] usb 1-1: new high-speed USB device number 2 using dwc_otg
[   57.030767] Indeed it is in host mode hprt0 = 00001101
[   57.052673] systemd[1]: Running with unpopulated /etc.
[   57.065863] systemd[1]: Set hostname to <ramdisk-9e22b8e6>.
[   57.073450] systemd[1]: Initializing machine ID from random generator.
[   57.232460] systemd[1]: Populated /etc with preset unit settings.
[   57.261400] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
[   57.270409] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   57.281317] uart-pl011 3f201000.serial: no DMA platform data
[... cut ...]
[   57.780876] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   57.791896] smsc95xx v1.0.5
[   57.846940] systemd-journald[125]: Received request to flush runtime journal from PID 1
[   57.897506] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:42:6c:fe
[   60.374213] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[   60.443953] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   60.453137] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
```

darkdrgn2k, 2018-03-01:
also running "init" from console seem to work without any unexpected errors

```
root@ramdisk-efc10622:/# ./init

Starting Ramdisk System

mount: proc is already mounted or /proc busy
       proc is already mounted on /proc
Checking for configuration files on mmcblk0p1
Applying configurations from /dev/mmcblk0p1: /mnt/conf.d/30-mesh-workshop.tar.gz
Executing configurations from /dev/mmcblk0p1: /mnt/conf.d/30-swap-filesystem.sh
[  107.832498] Adding 6143996k swap on /dev/mmcblk0p2.  Priority:-1 extents:1 across:6143996k SSFS
Executing configurations from /dev/mmcblk0p1: /mnt/conf.d/35-copy-dockerfiles.sh
Checking for configuration files on mmcblk0p2
mount: unknown filesystem type 'swap'
Error: mount /dev/mmcblk0p2 to /mnt as read-only failed
Error: processing of configuration files on mmcblk0p2 failed
Trying to run as user instance, but $XDG_RUNTIME_DIR is not set.
root@ramdisk-efc10622:/#
```

hamishcoleman, 2018-03-01:
I really do need you to give me the exact git commitid from this repository that you are using - if you are not using this repository to build things, I cannot use this repository to even try to replicate your issue.

Also, I'm somewhat confused now - what is your "regular" boot above?  that looks like it is booting fine to a ramdisk system.  The ramdisk system always boots from "/init", and it does not need to be specified (but if it is specified, it 

All that being said, I dont think I understand what you are trying to do...

darkdrgn2k, 2018-03-01:
Sorry reading back I realize I  was not clear without context.

The issue is with the custom configs not loading (the SCRIPT /init)

Regular is an unmodified kernel line
Vs the other one with is with init=/init

[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 elevator=deadline
 vs
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f00000                                                                                                                                            0  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 elevator=deadline init=/init

From what I understand during the boot process the [this script](https://github.com/hamishcoleman/debian-minimal-builder/blob/32d6aba1c716f7d816d36c5fe5541314dd6a178d/packages.d/_ALWAYS.fixup.add/init) is to run to install/apply the configs from /init.d which then execute [20confd.sh](https://github.com/hamishcoleman/debian-minimal-builder/blob/cbe60df03196b97f1d62f88ecc57a6193c30061e/packages.d/_ALWAYS.fixup.add/init.d/20confd.sh) that reads the conf.d scripts 

Problem is it look like /init is not running
ie "Starting Ramdisk System" does not appear anywhere nor do the configs from the sd card get applied

```
user@userPC:~/a/mesh-orange$ git describe
v0.6-76-g4587e71
```

hamishcoleman, 2018-03-01:
I think you missed my point - your "regular" build is clearly running "/init".  you do not need to specify it on the commandline, it is the default.

(I dont know why it wouldnt be working when you do specify it, but that just muddies the issue here)

So, currently, I still dont understand what you are trying to do - the /init script has run in your 'regular' boot - otherwise the hostname wouldnt be 'ramdisk-*'

hamishcoleman, 2018-03-01:
Can you duplicate your issue using the boards/qemu_i386 test build?

That is the simplest and fastest way to debug things

darkdrgn2k, 2018-03-01:
the config from the sd card was not applied

PS: this exact setup works fine on a pi3

darkdrgn2k, 2018-03-01:
how do you replicate the files on a boot partition in quemm?

```
mount /dev/mmcblk0p1  /boot
root@ramdisk-efc10622:/# cd boot
root@ramdisk-efc10622:/boot# cd conf.d
root@ramdisk-efc10622:/boot/conf.d# ls
30-mesh-workshop.tar.gz  30-swap-filesystem.sh  35-copy-dockerfiles.sh
root@ramdisk-efc10622:/boot/conf.d# 
```

hamishcoleman, 2018-03-01:
There is no difference between the pi2 and pi3 in this part of the system.

Are you using the exact same sdcard to boot both the pi2 and pi3?  If not, can you try?

hamishcoleman, 2018-03-01:
You will need to add a persistant disk to the qemu test command - there is an example of how to do this in the debian builder (see https://github.com/hamishcoleman/debian-minimal-builder/blob/master/test/Makefile#L54)

darkdrgn2k, 2018-03-01:
Same SD card on a pi 3
notice the host-name is different (stored in the sd card's config)

```[    0.000000] Linux version 4.9.35-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1014 SMP Fri Jun 30 14:47:43 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3d800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 14 pages/cpu @bcef8000 s25600 r8192 d23552 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 251720
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 elevator=deadline
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 932464K/1015808K available (7168K kernel code, 484K rwdata, 2012K rodata, 1024K init, 778K bss, 75152K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbe000000   ( 992 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c79014   ( 485 kB)
[    0.000000]        .bss : 0x80c7b000 - 0x80d3da64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000026] Switching to timer-based delay loop, resolution 52ns
[    0.000338] Console: colour dummy device 80x30
[    0.001225] console [tty1] enabled
[    0.001280] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001350] pid_max: default: 32768 minimum: 301
[    0.001705] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001748] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002844] Disabling cpuset control group subsystem
[    0.003030] CPU: Testing write buffer coherency: ok
[    0.003100] ftrace: allocating 22390 entries in 66 pages
[    0.055476] CPU0: update cpu_capacity 1024
[    0.055543] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.055642] Setting up static identity map for 0x100000 - 0x100034
[    0.057854] CPU1: update cpu_capacity 1024
[    0.057862] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.058572] CPU2: update cpu_capacity 1024
[    0.058578] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.059245] CPU3: update cpu_capacity 1024
[    0.059252] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.059334] Brought up 4 CPUs
[    0.059507] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.059536] CPU: All CPU(s) started in HYP mode.
[    0.059562] CPU: Virtualization extensions available.
[    0.060322] devtmpfs: initialized
[    0.071166] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.071479] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.071541] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.072103] pinctrl core: initialized pinctrl subsystem
[    0.073024] NET: Registered protocol family 16
[    0.075180] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.083916] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.083964] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.084139] Serial: AMBA PL011 UART driver
[    0.086044] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.086590] uart-pl011 3f201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.086954] irq: no irq domain found for /soc/aux@0x7e215000 !
[... 150 lines cut ...]
[   50.262492] systemd-journald[134]: Received request to flush runtime journal from PID 1
[   52.119033] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[   52.189042] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   52.198221] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   52.289750] usbcore: registered new interface driver brcmfmac
[   52.391179] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[... cut ...]
[   52.854943] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

Debian GNU/Linux 9 bloor ttyS0

bloor login: root
Password:
Linux bloor 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
root@bloor:~#
```

darkdrgn2k, 2018-03-01:
Sorry missed this part so in case its relavent
booting with `init=/bin/sh` and `init=/bin/bash` did NOT work took me to login prompt on both pi 2 and 3

also all the echos in the init scripts don't seem to ever display anywhere
do they go somewhere or is the syslog that is not running

hamishcoleman, 2018-03-01:
Thanks for testing with the init=/bin/bash - I definitely think there are two problems here, and one of them is less relevant to you (the one where init= is not working for some reason)

The echos in the init scripts /should/ be going to the serial console - as can be seen when you run the test environments.

I would expect them to appear some time just after
    [   49.010512] Freeing unused kernel memory: 1024K (80b00000 - 80c00000)

Perhaps it is a kernel module load order issue, perhaps it is something that the raspberry pi foundation changed upstream.  Did this issue used to work on older builds?  And again, I have to ask you to test this using just the mesh-orange repository - without any addons, so I can be sure what I am looking at

darkdrgn2k, 2018-03-01:
I just confirmed that this is the mesh-orange repo not ben's build.

the only "addon" is the conf.d on the vfat partition  that does not seem to work anyway.

We never had a Raspberry pi 2 to test with, i just got one few weeks ago.

We also learned that the workshop we will be conducting will be  on rpi 2 not 3s as previously hoped. So this is a first kick at the can for PI2 issue.

hamishcoleman, 2018-03-01:
What do you mean that that is an addon? - the default build out of the mesh-orange repo does have a conf.d

Can you tell me the commit ID of the checkout that built your issue, without that I cannot reproduce your environment.   If you have added any files to that after it is built, can you tell me what files and what contents.

It would be good if you can confirm if your issue is reproducible with the boards/qemu_i386 and boards/qemu_armhf test environments as I do not have a hardware test bed setup to test anything.

darkdrgn2k, 2018-03-01:
user@userPC:~/a/mesh-orange$ git rev-parse HEAD
4587e715ea3b434286c8d209948050f51414448d

The only thing that is done after the mesh-orange image is created the files for the workshop are added to the /conf.d of the vfat partition 

https://github.com/benhylau/mesh-workshop/releases/download/v0.1/confd-0.1.tar.gz 

This is the part that doesn't seem to work. It creates the swap parition, and such.

Specifically I copied the content of the "/conf.d/bloor"  folder from the above mentioned tar.gz into the /conf.d of the flashed sd card

darkdrgn2k, 2018-03-01:
guess i should also mention

user@userPC:~/a/mesh-orange/debian$ git rev-parse HEAD
cbe60df03196b97f1d62f88ecc57a6193c30061e

hamishcoleman, 2018-03-01:
Subdirectories in conf.d are not supported. 

Also:
Dont know what all those strange dotfiles are doing there either.  It really looks to me like you are trying to force the system to be something that it is not.  Perhaps you have the wrong tools.

hamishcoleman, 2018-03-01:
multiple hosts on the same network are not really something you want to do either - which is why the system generates a hostname

hamishcoleman, 2018-03-01:
Just for your information, a better command for identifying the commit you are on is "git describe --dirty --always"

benhylau, 2018-03-01:
Let me clarify, what @darkdrgn2k meant is the assets are copied over from the downloaded archive like this:

```
cp conf.d/bloor/* /mnt/to/BOOT/conf.d/
```

After that copy, the SD card will look like:

```
/BOOT/conf.d/30-mesh-workshop.tar.gz
             30-swap-filesystem.sh
             35-copy-dockerfiles.sh
```

There are no subdirectories. Perhaps it is easier to simply zip up a custom **/etc/hostname** into a tar.gz and observe that it does not get applied on a Raspberry Pi 2, using a vanilla build from here. A qemu test is probably not a goos test, since this seems Raspberry Pi 2 specific. The conf.d works both on Orange Pi Zero and Raspberry Pi 3, having this work on qemu doesn't tell us much.

hamishcoleman, 2018-03-01:
It is a simple request - and one that makes sense for all reported issues: "Can you replicate this issue in the provided test environment".

Yes, not all the parts are there, which is why I have pointed people both at the test environment  in the debian-minimal-builder and at the commandline it uses to attach the persistent storage (patches welcome!)

I just used the test environment and was able to replicate this issue.

The issue I found is that the kernel block drivers are configured as modules in some (but not all) kernels being used.

I do not currently have any hardware test environment, so I will not be able make a fix until I build one.

benhylau, 2018-03-01:
@hamishcoleman I reread our previous chat logs and finally realized what you meant now, with the debian-minimal-builder line you linked:

>2018-02-06 03:20:00     benhylau        I didn't see the /dev/vda in qemu, do I have to run the "make -C boards/qemu_i386 test" differently?
...
2018-02-07 00:36:24     hamish  The debian minimal builder has a test environment that has a persistant disk storage file attached.  If you are using the mesh-orange boards/qemu_i386 target, you will not have a persistant block image - you could port it over

So a patch like this? https://github.com/benhylau/mesh-orange/commit/25060e8f06e087bf96f35ee7887910a9ab4edaff

How do I manually access the persistent storage to put these in before the qemu envrionment boots up?

```
/persistent.storage/conf.d/30-mesh-workshop.tar.gz
                           30-swap-filesystem.sh
                           35-copy-dockerfiles.sh
```

darkdrgn2k, 2018-03-01:
> The issue I found is that the kernel block drivers are configured as modules in some (but not all) kernels being used.

So we just need a insmod during the "/init"  file or similar?

if it helps
```
root@ramdisk-f52a57d5:~# lsmod
Module                  Size  Used by
bcm2835_gpiomem         3940  0
uio_pdrv_genirq         3923  0
fixed                   3285  0
uio                    10204  1 uio_pdrv_genirq
ip_tables              13161  0
x_tables               20578  1 ip_tables
ipv6                  408971  22
root@ramdisk-f52a57d5:~#
```

also is there a way to edit the source of the roofs ramdisk image directly? without having to re-compile the whole image?  i tried mounting initrd but had no luck (im assuming thats the root fs)

darkdrgn2k, 2018-03-02:
**SERIAL OUTPUT**
I found that there was allot more logging being outputted on the HDMI console then on the serial port. 

I found that adjusting the `cmdline.txt` file as follows produces all output to serial port while keeping the HDMI completely clear

```
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 elevator=deadline
```
(still no output until the kernel is extracted but much more detailed after that!)

**Pi2 Boot Issue**

To test I modified the init script to include   `/bin/bash` as the first command.
Once booted into bash i was able to run the individual commands without an issue
I rebooted and issues an exit command, it continued booting without an issue

When I removed that line and booted again it did not work.
However with the new console modification i was able to see this
```
Starting Ramdisk System 

[   56.778366] random: fast init done
[   56.859432] mmc0: host does not support reading read-only switch, assuming write-enable
[   56.868982] systemd[1]: System time before build time, advancing clock.
[   56.879681] mmc0: new high speed SDXC card at address aaaa
[   56.886307] mmcblk0: mmc0:aaaa SL64G 59.5 GiB
[   56.890801] usb 1-1: new high-speed USB device number 2 using dwc_otg
[   56.897920] Indeed it is in host mode hprt0 = 00001101
[   56.905798] NET: Registered protocol family 10
[   56.915888] ip_tables: (C) 2000-2006 Netfilter Core Team
[   56.968743] systemd[1]: systemd 232 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN)
[   56.987005]  mmcblk0: p1
[   56.987802] systemd[1]: Detected architecture arm.
[   56.987858] systemd[1]: Running with unpopulated /etc.
```

I modified the script again to include a bit more information and a delay (probably to long at 50 seconds i think 5 or 10 would do) and it seemed to boot fine!

```
---SLEEPING---
[   56.823927] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:42:6c:fe
[   56.859297] mmc0: host does not support reading read-only switch, assuming write-enable
[   56.871393] mmc0: new high speed SDXC card at address aaaa
[   56.877811] mmcblk0: mmc0:aaaa SL64G 59.5 GiB
[   56.886316]  mmcblk0: p1
---AWAKE---  
--- RUNNING ----

Starting Ramdisk System

--- END ---
--- RUNNING ----
Checking for configuration files on mmcblk0p1
Applying configurations from /dev/mmcblk0p1: /mnt/conf.d/30-mesh-workshop.tar.gz
Executing configurations from /dev/mmcblk0p1: /mnt/conf.d/30-swap-filesystem.sh
[  107.146552] Adding 6143996k swap on /dev/mmcblk0p2.  Priority:-1 extents:1 across:6143996k SSFS
Executing configurations from /dev/mmcblk0p1: /mnt/conf.d/35-copy-dockerfiles.sh
--- END ---
--- RUNNING ----
--- END ---
--- RUNNING ----
[  107.272600] systemd[1]: System time before build time, advancing clock.
[  107.301407] NET: Registered protocol family 10
[  107.312054] ip_tables: (C) 2000-2006 Netfilter Core Team
[  107.330391] systemd[1]: systemd 232 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN)
[  107.349242] systemd[1]: Detected architecture arm.
[  107.354097] systemd[1]: Running with unpopulated /etc.

```

darkdrgn2k, 2018-03-02:
Adding a  `sleep 5 ` to the init script allows the conf files to be loaded correctly.

hamishcoleman, 2018-03-03:
@benhylau yes something like that patch, except that the armhf kernel doesnt seem to work with the virtio block devices, so that is best left out until investigation has been done.  And to match the rest of the boards, the persistant file should probably be in the board's $(BUILD) directory so that it gets removed by a 'make reallyclean'

There are many ways to access the contents of these files before booting the qemu (including mtools, loopback mounts and libguestfs) but as I have said previously, the simplest way is probably to boot up the qemu instance and scp your files in.  If you need to automate placing these files, then your best bet is to look at how the mesh-orange project works (it is probably more than 40% a wrapper around the tools needed to manipulate disk images)

hamishcoleman, 2018-03-03:
@darkdrgn2k the issue I found (which had the same symptoms as the issue you were finding) just needed some insmod, yes.  Your lsmod makes it clear that there is a different issue with the raspberry pi hardware.

The initrd is a compressed cpio archive, so you can extract its contents and re-archive it, but that is not going to be significantly faster.  So, it depends on what you want to edit in the ramdisk image.

I have just committed some changes to the debian builder to allow it to keep better track of dependencies, so it can easily re-build the image when it detects files changed (see the PR #33)

With that applied, if the files you are changing are in one of the CONFIGDIRS (this is the debian-config/packages.d and debian/packages.d dirs in the mesh-orange project) then a "make clean" is not needed.  (It doesnt detect new files, just edits to existing ones - if you do add a new file, removing the .configdir.deps file will cause it to regenerate the dependencies and find your new file - this is harder to automate quickly, however)

hamishcoleman, 2018-03-03:
I dont know why enabling the kernel debugger (with the kgdboc= kernel commandline param) would be adding more non-debugging output to the serial console, so that is probably a bad idea  - especially since you are not performing kernel debugging!

There is some strangeness with the pi2 and pi3 hardware config - they changed which SoC hardware block was used (and thus the driver) for the MMC to allow adding the WiFi, they also changed the hardware block used for serial console (and again, the driver) to allow using the bluetooth.  So, that probably explains the difference seen between the two hardware types - the rpi2 device driver probably is slightly slower at enumerating.

Based on @darkdrgn2k 's clear testing notes, I have committed a script to the debian builder that watches for block devices and waits up to 20 seconds for them to appear (but will exit sooner if they appear sooner)

darkdrgn2k, 2018-03-05:
is the change in /init i may be doing something wrong doesn't seem to work.

hamishcoleman, 2018-03-05:
There are at least 4 changes in different places.  I would have to go through every commit on that branch and cherrypick to find specifics, so your best bet is to just fetch the whole branch

darkdrgn2k, 2018-03-05:
just to confirm.. mesh orange is pined to a specific commit right?

hamishcoleman, 2018-03-05:
Yes - the submodule points at a specific commit.  However, if you pull the branch from the PR, it adjusts that pointer to a new commit, bringing in these changes.

darkdrgn2k, 2018-03-11:
Tried it this morning seem to work fine now thanks.

closed 2018-03-11 by hamishcoleman