tomeshnet/mesh-orange #37: Early boot output over serial - orange pi zero

darkdrgn2k, 2018-03-11:
The current orange pi build does not output anything to serial after "Starting kernel ..."

After poking at it i found the issue is probably in zImage
replacing the stock zImage from Armbian yields

```
Starting kernel ...

Uncompressing Linux... done, booting the kernel.
```

hamishcoleman, 2018-03-11:
Are you saying that the current build crashes before finishing booting, or that there are messages not seen on the output?

I built an image from v0.6-91-gb352b7972309 and had no issues (there is a significant delay between "starting kernel" and the next line - as usual)

<code>
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.13.0 (hamish@x230i5) (gcc version 6.3.0 20170516 (Debian 6.3.0-18)) #25 SMP Tue Sep 12 14:03:50 HKT 2017
</code>

darkdrgn2k, 2018-03-11:
You are correct

In armbian there are a few more lines between starting kernel and 0.0000

They are not important  unless you have issues with initrd.  The early error messages will not show

Not a huge bug.. Just though is document that I isolated it to  zImage (using armbian zImage produces the missing output)

hamishcoleman, 2018-03-11:
ah, in this case it is a simple one:  Its a kernel config option to set the amount of messages on the output add the kernel cmdline param "loglevel=7" and you should see some extra messages.

I added that setting into the kernel config to go into the next kernel build.