tomeshnet/mesh-orange #42: Failure to verify licheepizero.us's certificate

benhylau, 2018-03-28:
```
...
/home/travis/build/tomeshnet/mesh-orange/mk/common.mk:48: warning: ignoring old commands for target `clean'
mkdir -p build
wget -O build/zero_imager.zip https://licheepizero.us/downloads/zero_imager.zip
--2018-03-28 17:02:03--  https://licheepizero.us/downloads/zero_imager.zip
Resolving licheepizero.us (licheepizero.us)... 89.33.16.163
Connecting to licheepizero.us (licheepizero.us)|89.33.16.163|:443... connected.
ERROR: cannot verify licheepizero.us's certificate, issued by ‘/C=US/O=Let's Encrypt/CN=Let's Encrypt Authority X3’:
  Issued certificate has expired.
To connect to licheepizero.us insecurely, use `--no-check-certificate'.
make[1]: *** [build/zero_imager.zip] Error 5
make[1]: Leaving directory `/home/travis/build/tomeshnet/mesh-orange/boards/sun8i-v3s-licheepi-zero'
make: *** [image] Error 2
```

hamishcoleman, 2018-03-29:
Not much I can do about this one.  Clearly their lets-encrypt automation has failed.

Turning off cert checking doesnt sound like a good plan either

hamishcoleman, 2018-03-29:
I've bugged the site owner, but I'll see what else I can think of as well

hamishcoleman, 2018-03-29:
The website was not dead, so we were able to get that fixed - builds are passing now

closed 2018-03-29 by hamishcoleman