tomeshnet/mesh-orange #46: Need list of firmware

hamishcoleman, 2018-04-05:
In order to improve the way we are handling the firmware, I need to remove most of the firmware files that are currently installed.  To ensure that we still have the firmware needed to support for all the expected hardware, I need to ensure some files are added in - but I dont have access to most (all?) of the required hardware to confirm the needed files.

We need to compile a list of supported hardware and the matching firmware.

@benhylau, @darkdrgn2k  can you both add all the details you have for this?

#27 and #22 look like they might help

Current list of hardware I know about:
* (orange-pi-zero) xradio
* (raspberry-pi-3) brcmfmac43430-sdio
* (usb) TOP-GS07 RT5572
* (usb) Ath9k
* (usb) Some new USB stick Ben has trialed

Still need exact firmware file names for most of those though.

darkdrgn2k, 2018-04-05:
Ill start work on that however a repository of some information can be found [here](https://github.com/phillymesh/802.11s-adapters)

Maybe i can include the firmware information in that repo as we test.

benhylau, 2018-04-06:
I used `rt2800usb ath9k_htc rtl8192cu`

hamishcoleman, 2018-04-06:
and what firmware files are needed by each of those?

benhylau, 2018-04-06:
How do I check that?

hamishcoleman, 2018-04-07:
* boot up a system
* rename the /lib/firmware dir to something else

* insert the usb stick
* look at the output of dmesg for which firmware files it is trying to load
* rename only those files back into the correct /lib/firmware dir (and make note of them)
* re-insert the usb stick
* check to see if it is working.
* if not, repeat for whatever file it is asking for now
* it it is working, you now have a list of the required firmware

darkdrgn2k, 2018-04-07:
Most of my unique dongles are scattered into the wind at the moment as i been brining the TP 722s for the workshop

But the two main ones are
```
TL-WN722N V1
ath9k_htc
htc_9271.fw

Toplinkst TOP-GS07
rt2800usb
rt2870.bin
```

I think the new chinees one is the same as the topink but i can only test it on a orange pi.

benhylau, 2018-04-13:
I tried those two @darkdrgn2k posted and have the same results. Also, I tried the new [TOP-4M02](http://www.toplinkst.com/html_products/80211abgn-RT5572N-24GHz5GHz-dual-band-300Mbps-usb-wireless-wifi-embedded-module-56.html) and it uses the same file as the TOP-GS07.

I do not have the one that is `rtl8192cu` anymore. It was borrowed from the library and worked very poorly so we ended up returning it.

hamishcoleman, 2018-06-03:
The firmware package updates needed to support the rpi3b+ mean that the ralink firmware is no longer available - this needs some work to address.