tomeshnet/mesh-orange #47: Raspbery PI 3b+ does not boot

darkdrgn2k, 2018-04-06:
No output on serial on boot
But  looks like there is a new dtb 
```
 bcm2710-rpi-3-b-plus.dtb
```

darkdrgn2k, 2018-04-06:
Simply copying over the bcm2710-rpi-3-b-plus.dtb did not seem to work

Power on and about 10-15 seconds later the  power light flashing as
4 long blinks, 3 long blinks, 1 long blink (like 3 second long blink)

hamishcoleman, 2018-04-06:
This is not really unexpected.

Unfortunately, my local suppliers of Raspberry pi 3B+ have been out of stock since it was launched.

darkdrgn2k, 2018-04-06:
any  suggestions what needs to be added to get it workigng?

hamishcoleman, 2018-04-06:
perhaps start with a working raspbian sdcard and slowly add the bits from the tomesh image?

darkdrgn2k, 2018-04-06:
Just wanted to make sure that there is no "compiling" that happens during the rapsberry pi image build when it comes to the boot directory.

darkdrgn2k, 2018-04-06:
Got it booting

1 New file Needed
bcm2710-rpi-3-b-plus.dtb

Two updated
start_cd.elf and  fixup_cd.dat

hamishcoleman, 2018-04-07:
Which image version did you test with?  I would not have expected any differences with the start_cd.elf and fixup_cd.dat as they are downloaded at the time the image is built.   So, I would expect a freshly built image to only need the dtb!

darkdrgn2k, 2018-04-07:
I think i had a thought like that but it must have gotten lost amongst todays chaos.

I think you are totally right. My image was an older one (bens release for weeks ago) that was done probably before the PI3b+ was released.

In that case it looks like were just missing  bcm2710-rpi-3-b-plus.dtb and we are good.

I will test by rebuilding when i have a moment but i think its a safe bet to say we just need that one file to make it work!

darkdrgn2k, 2018-04-07:
Confirmed with today's MASTER branch

Compiled copied over the  bcm2710-rpi-3-b-plus.dtb to boot with no luck
Replaced  start_cd.elf and fixup_cd.dat from raspbian 2018 and it booted

hamishcoleman, 2018-04-07:
If that is the case, then the raspbian people must have stuffed up their packaging.

To confirm, can you look in the apt.cache dir and tell me what the raspberrypi-bootloader_* filename found there is?

That firmware error is unrelated to booting, but worth noting for that other ticket.

hamishcoleman, 2018-04-07:
looking at http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ shows lots of raspberrypi-bootloader versions

- version 1.20170703-1 is the latest shown in the raspbian Packages list,
- version 1.20180328-1 is the latest in that directory

So much for all that effort I just put into a generic deb downloader - that only works if the bozos running the repo keep it up to date..

darkdrgn2k, 2018-04-07:
Cant seem to find an apt.cache on the new raspbian image

hamishcoleman, 2018-04-07:
in the git checkout

darkdrgn2k, 2018-04-07:
Ahh on mesh orange

```
drwxrwxr-x  4 user user 4.0K Apr  7 09:39 .
drwxrwxr-x 15 user user 4.0K Apr  7 09:39 ..
-rw-rw-r--  1 user user  235 Apr  7 09:39 apt_config
drwxrwxr-x  3 user user 4.0K Apr  7 09:35 etc
-rw-r--r--  1 root root 853K Aug 16  2017 firmware-atheros_1%3a0.43+rpi6_all.deb
-rw-r--r--  1 root root 1.7M Aug 16  2017 firmware-brcm80211_1%3a0.43+rpi6_all.deb
-rw-r--r--  1 root root  46K Aug 16  2017 firmware-ralink_1%3a0.43+rpi6_all.deb
-rw-rw-r--  1 user user  668 Apr  7 09:39 print_uris.txt
-rw-rw-r--  1 user user 3.4M Jul  3  2017 raspberrypi-bootloader_1.20170703-1_armhf.deb
-rw-rw-r--  1 user user  31M Jul  3  2017 raspberrypi-kernel_1.20170703-1_armhf.deb
drwxrwxr-x  4 user user 4.0K Apr  7 09:35 var
```

darkdrgn2k, 2018-04-07:
Wifi and Ethernet not working on pi3b+ also
(they where upgraded to gigabit eth and AC wifi)

Information from booted Raspbian 2018
```
  *-network:0 DISABLED      
       description: Wireless interface
       physical id: 2
       logical name: wlan0
       serial: b8:27:eb:13:76:3a
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=brcmfmac driverversion=7.45.154 firmware=01-4fbe0b04 multicast=yes wireless=IEEE 802.11
  *-network:1
       description: Ethernet interface
       physical id: 3
       logical name: eth0
       serial: b8:27:eb:46:23:6f
       size: 1Gbit/s
       capacity: 1Gbit/s
       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=lan78xx driverversion=1.0.4 duplex=full ip=192.168.40.214 link=yes multicast=yes port=MII speed=1Gbit/s
```

hamishcoleman, 2018-04-07:
The correct kernel will have the correct device drivers - unfortunately the raspbian repos are not publishing the 'correct' kernel.

The wifi may need firmware, which is intended to be fixed in a different way, but might depend on how mucked up the raspbian packaged are

hamishcoleman, 2018-04-07:
can you try applying this patch to the tomesh-orange repo, rebuilding the image and add the new dtb.  This may not affect the wifi function or the ethernet, but I want to test booting

```
diff --git a/apt.sources/raspbian.list b/apt.sources/raspbian.list
index 2f4851197a49..4a4535706540 100644
--- a/apt.sources/raspbian.list
+++ b/apt.sources/raspbian.list
@@ -4,4 +4,4 @@
 # This is used to download the raspberry pi bootloader blob and kernel images
 #
 
-deb http://archive.raspberrypi.org/debian/ jessie main
+deb http://archive.raspberrypi.org/debian/ stretch main
```

darkdrgn2k, 2018-04-07:
No dice. Same issue (flashing lights no boot.)

hamishcoleman, 2018-04-08:
Hrm.

In that case... know any way I can get a rpi3b+ in Hong Kong without waiting for 1 or 2 months for stock availability? ...

darkdrgn2k, 2018-04-08:
They seem to be getting out of stock in Canada to now.  Found [these](https://www.canakit.com/raspberry-pi-3-model-b-plus.html?cid=cad&src=raspberrypi&src=raspberrypi) guys but prob wont ship to hong kong :/

darkdrgn2k, 2018-04-08:
i don't blame you our prices are almost double :P

well until then if you want me to try something let me know

hamishcoleman, 2018-04-08:
It seems there might be some regulatory issue getting them in hong kong: "Compliance Approval Pending - stock due April 2018"

I'll see if I can cook up some more tests and will let you know if I think of anything - thanks

benhylau, 2018-06-08:
Thanks for fixing and verifying this @hamishcoleman and @darkdrgn2k. This is eligible for [ssbc-grants-scuttlemesh](https://github.com/benhylau/ssbc-grants-scuttlemesh) :)

closed 2018-06-08 by benhylau