tomeshnet/mesh-orange #48: Non-ramdisk images with data persistence across power cycles

benhylau, 2018-05-08:
Building both a ramdisk and non-ramdisk .img has been discussed between @hamishcoleman @darkdrgn2k and I.

benhylau, 2018-06-28:
@hamishcoleman Would like some clarification to how you envision the partitioning to work. In previous conversations we discussed building two .img artifacts, one with **ramdisk** as it is now, and a new one that has a **sdcard root**.

From [boards/common-partitions.mk](https://github.com/tomeshnet/mesh-orange/blob/master/boards/common-partitions.mk#L20-L22) it seems we will have three partitions:

```
PART1_SIZE_MEGS ?= 1000
#PART2_SIZE_MEGS ?= 1000 # reserved for swap
#PART3_SIZE_MEGS ?= 1000 # ext2 root
```

How will these be set for **ramdisk** and **sdcard root**?

Am I correct to assume that having `PART2_SIZE_MEGS` enabled will solve the space problem we have in https://github.com/tomeshnet/mesh-orange/issues/52? Since the swap fs I create via conf.d scripts will now be expanded into a ramdisk that is backed by the swap space? Or does `debian.stretch.armhf.cpio` expansion occur _before_ the swap space is enabled which means we'd be out of space before `PART2` kicks in?

hamishcoleman, 2018-06-28:
The cpio expansion happens way before anything else.  It happens before the kernel has even finished booting.  There is no way to run any scripts or mount any swapspace before the expansion happens.

As stated, I dont know that I can answer your question about partitions - I have not finished writing the code, so it might change that file completely, it would probably not be a simple case of uncommenting those lines (as that would force building one or the other style of image on any given build invocation - which is not ideal)

However, yes, I expect there to be three partitions, one for boot, one for swap and one for root.

darkdrgn2k, 2018-06-28:
> The cpio expansion happens way before anything else. It happens before the kernel has even finished booting. There is no way to run any scripts or mount any swapspace before the expansion happens.

Unless you rip out all the packages, and the install them at boot time....

benhylau, 2018-06-28:
@darkdrgn2k but that's all the stuff in the `conf.d` directory right? That's not part of the cpio I think?

darkdrgn2k, 2018-06-28:
yes, but then you are also risking running out of room on the fat partition