tomeshnet/mesh-orange #52: RPI 3B+ not booting after adding some packages

darkdrgn2k, 2018-06-16:
When using http://github.com/benhylau/mesh-router-builder to build the mesh-workshop profile the RPI 3B+ will no longer boots and complains about EOF on the ram disk. However using the same builder without the "mesh-workshop" extras (default profile) it boots fine (just like stock mesh-orange).

It seems the only difference between the two are package that are added.

I have included the error copied form serial console durring boot, and the difference between Default and Mesh-Workshop profiles.

```
 100.660858] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[  100.666492] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[  100.673149] of_cfs_init
[  100.675730] of_cfs_init: OK
[  100.679199] RAMDISK: lzma image found at block 0
[  100.706675] RAMDISK: EOF while reading compressed data
[  100.706683] unexpected EOF
[  100.716473] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[  100.724084] Please append a correct "root=" boot option; here are the available partitions:
[  100.732593] 0100            4096 ram0
[  100.732598]  (driver?)
[  100.738788] 0101            4096 ram1
[  100.738792]  (driver?)
[  100.744996] 0102            4096 ram2
[  100.745001]  (driver?)
[  100.751190] 0103            4096 ram3
[  100.751195]  (driver?)
[  100.757401] 0104            4096 ram4
[  100.757406]  (driver?)
[  100.763611] 0105            4096 ram5
[  100.763615]  (driver?)
[  100.768089] mmc1: new high speed SDIO card at address 0001
[  100.775376] 0106            4096 ram6
[  100.775381]  (driver?)
[  100.781587] 0107            4096 ram7
[  100.781592]  (driver?)
[  100.787780] 0108            4096 ram8
[  100.787784]  (driver?)
[  100.791254] usb 1-1: new high-speed USB device number 2 using dwc_otg
[  100.791370] Indeed it is in host mode hprt0 = 00001101
[  100.805737] 0109            4096 ram9
[  100.805742]  (driver?)
[  100.811948] 010a            4096 ram10
[  100.811952]  (driver?)
[  100.818229] 010b            4096 ram11
[  100.818233]  (driver?)
[  100.824527] 010c            4096 ram12
[  100.824532]  (driver?)
[  100.830808] 010d            4096 ram13
[  100.830812]  (driver?)
[  100.837113] 010e            4096 ram14
[  100.837117]  (driver?)
[  100.843410] 010f            4096 ram15
[  100.843418]  (driver?)
[  100.849702] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[  100.858091] CPU: 2 PID: 1 Comm: swapper/0 Not tainted 4.14.34-v7+ #1110
[  100.864799] Hardware name: BCM2835
[  100.868275] [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
[  100.876142] [<8010c240>] (show_stack) from [<807840a4>] (dump_stack+0xd4/0x118)
[  100.883570] [<807840a4>] (dump_stack) from [<8011dc34>] (panic+0xf0/0x274)
[  100.890555] [<8011dc34>] (panic) from [<80b014f8>] (mount_block_root+0x1e8/0x2b8)
[  100.898154] [<80b014f8>] (mount_block_root) from [<80b017e8>] (mount_root+0x12c/0x134)
[  100.906193] [<80b017e8>] (mount_root) from [<80b01990>] (prepare_namespace+0x1a0/0x1e8)
[  100.914320] [<80b01990>] (prepare_namespace) from [<80b010ac>] (kernel_init_freeable+0x2a8/0x2bc
```

Difference in build
```
vagrant@stretch:~/mesh-orange-images/mesh-orange/mesh-workshop/debian-config$ diff debian.stretch.multistrap ../../default/debian-config/debian.stretch.multistrap 
2c2
< bootstrap=Stretch Docker Local
---
> bootstrap=Stretch Local
17d16
< # TODO get selected mesh router from node-profiles
19,27d17
< 
< [Docker]
< # TODO add keyring
< source=https://download.docker.com/linux/debian
< suite=stretch
< omitdebsrc=true
< components=nightly
< 
< packages= docker-ce
```

```
vagrant@stretch:~/mesh-orange-images/mesh-orange/mesh-workshop/debian-config$ diff packages.txt  ../../default/debian-config/packages.txt 
5,14d4
< # common ssl certificates
< ca-certificates
< 
< # mdns
< avahi-daemon
< libnss-mdns
< 
< # networking tools
< curl
< iperf3
16,18d5
< netcat
< 
< # text editor
```

hamishcoleman, 2018-06-17:
This looks like a classic case of the built image being too big for the RAM available.

Can you paste the entire boot log?  The initrd unpacking and out-of-memory errors occur much earlier in the boot log.

What are the sizes of the built .img files in all your tested cases?  (also, if possible, what are the sizes of the initrd files in all cases too)

I really dont know what all these other repos contain or are doing.  Do you have a minimal testcase patch to the mesh-orange repo that replicates this issue?

hamishcoleman, 2018-06-17:
That log pretty much confirms the theory that the image is too large - these log lines:
<pre>
[    0.160020] Trying to unpack rootfs image as initramfs...
[   98.764075] rootfs image is not initramfs (write error); looks like an initrd
[   98.764198] /initrd.image: incomplete write (4096 != 112347770)
</pre>
I think can occur for other reasons, but I have only seen them when there is too much to fit in RAM.

What are the sizes of the uncompressed debian images for all your tested builds? ("ls -l debian/build/debian.stretch.armhf.cpio" should show this)

darkdrgn2k, 2018-06-17:
```
vagrant@stretch:/tmp/mesh-orange$ ls -l debian/build/debian.stretch.armhf.cpio 
-rw-r--r-- 1 vagrant vagrant 307975680 Jun 16 04:10 debian/build/debian.stretch.armhf.cpio
```

The raspberry Pi 3b+ has 1 gig of ram
We ran these on the RPI3B boards for the workshop. I currently do not have a working 3B board to test with (@benhylau does it boot on the RPI3B?)

benhylau, 2018-06-17:
**On the Raspberry Pi 3B**

The `default` profile image that works:

```
vagrant@stretch:~$ sudo mount -o offset=$((512*2048)) -t vfat /vagrant/output/mesh-orange-images/raspberrypi3-0.8_default.img /mnt
vagrant@stretch:~$ ls -lh /mnt
total 73M
-rwxr-xr-x 1 root root  23K Jun 17 18:56 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root  24K Jun 17 18:56 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root  24K Jun 17 18:56 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root  23K Jun 17 18:56 bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root  51K Jun 17 18:56 bootcode.bin
-rwxr-xr-x 1 root root   65 Jun 17 18:56 cmdline.txt
drwxr-xr-x 2 root root  16K Jun 17 18:56 conf.d
-rwxr-xr-x 1 root root   66 Jun 17 18:56 config.txt
-rwxr-xr-x 1 root root  19K Jun 17 18:56 COPYING.linux
-rwxr-xr-x 1 root root 2.6K Jun 17 18:56 fixup_cd.dat
-rwxr-xr-x 1 root root  68M Jun 17 18:56 initrd
-rwxr-xr-x 1 root root 4.7M Jun 17 18:56 kernel7.img
-rwxr-xr-x 1 root root 1.5K Jun 17 18:56 LICENCE.broadcom
-rwxr-xr-x 1 root root 658K Jun 17 18:56 start_cd.elf
```

The `mesh-workshop` profile image that fails:

```
vagrant@stretch:~$ sudo mount -o offset=$((512*2048)) -t vfat /vagrant/output/mesh-orange-images/raspberrypi3-0.8_mesh-workshop.img /mnt
vagrant@stretch:~$ ls -lh /mnt
total 114M
-rwxr-xr-x 1 root root  23K Jun 13 05:26 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root  24K Jun 13 05:26 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root  24K Jun 13 05:26 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root  23K Jun 13 05:26 bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root  51K Jun 13 05:26 bootcode.bin
-rwxr-xr-x 1 root root   65 Jun 13 05:26 cmdline.txt
drwxr-xr-x 2 root root  16K Jun 13 05:26 conf.d
-rwxr-xr-x 1 root root   66 Jun 13 05:26 config.txt
-rwxr-xr-x 1 root root  19K Jun 13 05:26 COPYING.linux
-rwxr-xr-x 1 root root 2.6K Jun 13 05:26 fixup_cd.dat
-rwxr-xr-x 1 root root 109M Jun 13 05:26 initrd
-rwxr-xr-x 1 root root 4.7M Jun 13 05:26 kernel7.img
-rwxr-xr-x 1 root root 1.5K Jun 13 05:26 LICENCE.broadcom
-rwxr-xr-x 1 root root 658K Jun 13 05:26 start_cd.elf
```

darkdrgn2k, 2018-06-17:
Looks like its a size issue

any idea why since there seems to be more ram then needed?

@benhylau  any packages we can take out to see if it will work

benhylau, 2018-06-17:
Want to add that the previous version that worked (with both build profiles) have the following `initrd` sizes:

`default` -> `60M Apr 26 05:52 initrd`
`mesh-workshop` -> `98M Apr 26 06:48 initrd`

So it is the additional 11M from 98M to 109M, probably from ff3e59dc916b550b49c0531ff6198d6eac26f7fd and 1b762ae886af758d2137dce2c798de2a90edacf6. @hamishcoleman do these numbers look significant in any way?

darkdrgn2k, 2018-06-18:
Sounds like to me like we have a few options

- Finish the SD card root partition (progress?)
- Try to remove some packages we dont need
- Try to redruce the base image by removing file bloat man (ie files maybe)

Also i THINK there is a way to set the amount of ram the GPU takes in the pi. Ill try to look through the code (raspi-config) and see if it maybe is just a boot time argument.

hamishcoleman, 2018-06-18:
Before you go reading the raspi-config source, have a look at boards/raspberrypi2/config.txt - the line is "gpu_mem=16" (ref https://www.raspberrypi.org/documentation/configuration/config-txt/)

* sdcard root progress is slow because I have no time.
* removing packages sounds like a win
* reducing the base image is a simple process that has not been done for a long time now... but the gains will be far outweighed by the increase from adding all these new packages

benhylau, 2018-06-28:
>removing packages sounds like a win

I need all of them in the workshop image, so that's not really an option unfortunately.

>reducing the base image is a simple process that has not been done for a long time now... but the gains will be far outweighed by the increase from adding all these new packages

Agree.

darkdrgn2k, 2018-06-28:
Confirmed this is a lack of space issue. Stripped out most of the extra packages and finally got it to boot with a 24mb "free space"

Possibly two sets of kernel modules installed?
```
/lib/modules/4.14.34+/
/lib/modules/4.14.34-v7+/
```