tomeshnet/mesh-orange #53: Make a second node bootstrap from a working node

benhylau, 2018-07-23:
Had a conversation with @jedahan who brought up the use case of having one Pi with a blank SD be able to bootstrap off another, perhaps with a USB-SD adapter, or some sort of netboot mechanism. This could be related (but not exactly) to the OS upgrade flow. Tracking that question here to explore options.

darkdrgn2k, 2018-09-30:
Big problem is that nodes need a "bootloader" that is usually located on an SD card (or some other storage media)

Raspberry Pi can TFTP boot, but only if there is a boot loader on the sd card or i think UBS stick.

Without an SD card with a bootloader these devices are dead.
May be able to use SPI on some of these boards for the boot loader but not sure.

Even if you get TFTP setup you need a traditional network with DHCP and a TFTPd server to make it work.

hamishcoleman, 2018-09-30:
Modern raspberry pi devices have the network boot code in their ROM, and do not need any SDcard.  

However, with a running rpi and a USB-SD adaptor, the software could clone one sdcard to another.  The issue would be how to trigger it safely.

darkdrgn2k, 2018-10-01:
Interesting renetwork boot code.  I think i remember reading something somewhere about it a while back but never looked twice at it (so i dont know if its true or how to do it). But IIRC its not on by default so you still need an SD card to enable it.  However you still have to build a been an environment for it (tftpfd, configured dhcpd and a layer 2 wired network etc)

As for as USB-SD adapters,your essential making an "install disk" for a node. I would assume it could be just a plain USB stick and not even SD.

But i would like to ask what the end game here is.

If its to install one node of another by physically moving the SD card, could you not just boot a working RAMDISK  node with a working SD card then just replace the SD card with a bank one and run an OTA upgrade to write the boot code/files onto the SD card?

Seems like less work and already in our road map.

hamishcoleman, 2018-10-01:
You can read all about the built-in network boot at https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net.md - this is enabled by default on Raspberry Pi 3B+ and requires nothing more to get it started.  They even link to a fully worked example on how to get it all working on raspbian.

The environment needed for this is straightforward and nothing special - it is just not possible to configure the systemd-networkd built-in dhcpd to do this.  Yes, you will need an ethernet cable..

If you have a USB-SDcard adaptor then the sdcard made from it is the card that is inserted into the new PI - thus better than using a USB stick.

Regardless of what the endgame is, the idea that one node could clone itself seems like a good one.